michelsciortino/esphome-aurora-inverter #1: Compiling for esp8266 boards

NMBurdel, 2022-03-28:
Hi, i have an error when i try to compile the config.yaml for every 8266 boards. Can you help me ?
[logs_config_upload.txt](https://github.com/michelsciortino/esphome-aurora-inverter/files/8365610/logs_config_upload.txt)

,

sim-bax, 2023-02-24:
Did you solve it? I'm having the same problem

NMBurdel, 2023-02-24:
Yes i solved, try to modify in the ABBAurora.cpp the function ABBAurora::setup in this way:

void ABBAurora::setup(HardwareSerial &hardwareSerial, byte RXGpioPin, byte TXGpioPin, byte TXControllPin)
{
    TXPinControl = TXControllPin;

    pinMode(TXPinControl, OUTPUT);
    digitalWrite(TXPinControl, LOW);

    serial = &hardwareSerial;
    // OK per ESP32
    //serial->begin(19200, SERIAL_8N1, RXGpioPin, TXGpioPin, false, 500);
    // OK per ESP8266
    serial->begin(19200, SERIAL_8N1, SERIAL_FULL, TXGpioPin, false);
}

sim-bax, 2023-02-25:
it works!
Thank you!!

botr76, 2023-03-11:
Hello 
How to set up the serial port to serial1? I am getting an error with serial2 option.
    ABBAurora::setup(Serial1, RX, TX, TX_CONTROL_GPIO);     ok ?

How to set up the serial port 
#define LED 2               //GPIO02, the ESP32 internal led
#define RX 15               //GPIO14
#define TX 13               //GPIO27
#define TX_CONTROL_GPIO 12  //GPIO26
#define INVERTER_ADDRESS 2  //Default address; this can be modified on the inverter settings menu

I have a Wemos and with these settings, I can't find the inverter.

NMBurdel, 2023-03-11:
check the gpio scheme for your wemos board.. 
For example if you use a wemos D1 mini,
The TX pin is GPIO 1 and RX is GPIO 3 ecc...

botr76, 2023-03-11:
ok  thx  ... serial 1 ??

botr76, 2023-03-11:
#define LED 2               //GPIO02, the ESP32 internal led

#define RX 3               //GPIO14
#define TX 1               //GPIO27
#define TX_CONTROL_GPIO 12  //GPIO26
#define INVERTER_ADDRESS 2  //Default address; this can be modified on the inverter settings menu

I have a Wemos D1 mini. Is it okay now? I need confirmation because I've been trying to connect for several days."

botr76, 2023-03-13:
> check the gpio scheme for your wemos board.. For example if you use a wemos D1 mini, The TX pin is GPIO 1 and RX is GPIO 3 ecc...

Is the system on wemos d1 mini working for you?

sim-bax, 2023-03-14:
I am using nodeMCU with no success, but I think it is a problem of pin
In the next days I will try with D1mini, at least we can share our experiences

NMBurdel, 2023-03-14:
I have the nodeMCU and it works. 
@botr76 check also the TX_CONTROL_GPIO pin.

At the and, remember to change in the config.yaml file in order to select the correct board.
For my board:
...
esp8266:
   board: nodemcuv2
...

-----
What is you problem? compiling the code or communicating with the inverter in HA?

closed 2023-03-14 by NMBurdel

sim-bax, 2023-03-15:
My problem is communicating with the inverter
Can you share your GPIO configuration?
Maybe my error is there

botr76, 2023-03-15:
> Did you solve it? I'm having the same problem

I created a new topic...you're welcome to join