michelsciortino/esphome-aurora-inverter #4: Wemos D1 mini

botr76, 2023-03-15:
Can someone help with communication and adjusting the program to the Wemos D1 Mini? Below is my configuration, in the logs I have "inventer not found" error.

    serial = &hardwareSerial;
// OK per ESP32
    //serial->begin(19200, SERIAL_8N1, RXGpioPin, TXGpioPin, false, 500);
// OK per ESP8266
    serial->begin(19200, SERIAL_8N1, SERIAL_FULL, TXGpioPin, false);
    
    
--------------------
#define LED 2               //GPIO02, the ESP32 internal led

#define RX 3               //GPIO14
#define TX 1               //GPIO27
#define TX_CONTROL_GPIO 13  //GPIO26
#define INVERTER_ADDRESS 2  //Default address; this can be modified on the inverter settings menu

----------------
esphome:
  name: inwenter2
  platform: ESP8266
  board: d1_mini
  includes:
    - ABBAuroraStrings.cpp
    - ABBAuroraStrings.h
    - ABBAuroraEnums.h
    - ABBAurora.cpp
    - ABBAurora.h
    - InverterMonitor.h
 
  libraries:
    - "Wire"

sim-bax, 2023-03-22:
I had little improvement
in
InterverMonitor.h I have changed void setup()

  void setup() override
  {
    ESP_LOGD(TAG, "Setupping");
    pinMode(LED, OUTPUT);
    led_state = LOW;
    ABBAurora::setup(Serial, RX, TX, TX_CONTROL_GPIO); //it was Serial2
    inverter = new ABBAurora(INVERTER_ADDRESS);
    connection_status->publish_state(DISCONNECTED);
  }
  
  Slowly I'm getting the data

sim-bax, 2023-03-22:
The problem is that often I get this

17:20:31	[D]	[INVERTER_MONITOR:163]	
Inverter not conntected
17:20:46	[D]	[INVERTER_MONITOR:163]	
Inverter not conntected
17:21:01	[D]	[INVERTER_MONITOR:163]	
Inverter not conntected
17:21:16	[D]	[INVERTER_MONITOR:163]	
Inverter not conntected
17:21:31	[D]	[INVERTER_MONITOR:163]	
Inverter not conntecte

But sometimes it is connected

botr76, 2023-03-25:
Beautiful .. that you have communication at all
Can you show **what converter** you have, Which pins are you using? It's not going for me
I have 3 inverters and I still have to convert this program to read from three addresses.

sim-bax, 2023-06-09:
Ciao
Any update from your side? I had to stop to work on it, but now I'd like to start again

wergio, 2023-11-12:
anyone succeed with D1 mini? no success here with a Wemos D1 Mini 4.0.0...
with GPIO and 15-13 doesen't even boot, with 1-3 boots but "Inverter not connected"
I'm using this adapter with 5V vcc https://it.aliexpress.com/item/1005001621798947.html
maybe modifiing the code to use software serial would help?

wergio, 2023-11-15:
now is working with pin GPIO 1, 3, and 5 for control ! 
the solution is to move the esphome logger from uart0 to uart1 with this conf in yaml file
```
logger:
  hardware_uart: UART1
``` 
otherwise the ESP8266 serial logger writes to uart0 by default and interfere with inverter communication
you can always read the log with web logger from esphome component in home assistant

Rossigg, 2024-05-04:
Similar problem for me.
I use a NodeMCU with esp8266 connected to a TTL to RS485 converter. The major of time it returns the error "Inverter not connected" and some times it can connect to server, logs the reading, but not send to Home Assistant.
I use 1 and 3 pin because if I use 15-13 the device cant boot. I also tried to put URT1 into logger configuration but nothing happened.

Any suggenstion?