michelsciortino/esphome-aurora-inverter #8: Inverter not connected - how to do the cabling?

kevinholtkamp, 2023-10-16:
I'm getting the following logs from my esp:
```
[17:03:19][D][INVERTER_MONITOR:163]: Inverter not conntected
[17:03:19][W][component:204]: Component <unknown> took a long time for an operation (1.06 s).
[17:03:19][W][component:205]: Components should block for at most 20-30ms.
```
I'm using an ESP32 connected to our Aurora Inverter, using the default configuration and a Waveshare TTL->RS485 adapter:
[image]
[image]

Can anybody tell me where the issue is? I tried swapping cabled, and the inverter is set to be device 2.

In general I think it would be a good idea to have example images of how to connect the cables in the documentation.

Yaou26, 2024-07-17:
I have the same problem. I've been trying to get it to work for days. Can someone share their .yaml and inverter.h as well as their connection?

Birbodes, 2025-02-08:
The reason could be, that you switched the 120 Ohm Terminator resistor on. The rs485 to ttl converter is the "real" end of the daisy-chain an normally already has an 120 Ohms resistor between  the A/B Pins to terminate the daisy-chain.