michelsciortino/esphome-aurora-inverter #9: How to load this thing?

incarvr6, 2023-11-07:
How to load this thing? I keep getting error: src/ABBAurora.h:5:10: fatal error: Wire.h: No such file or directory
all files are in esphome folder.

wergio, 2023-11-10:
you must add in yaml file

  libraries:
    - Wire

after "includes"

incarvr6, 2023-11-10:
> you must add in yaml file
> 
> libraries: - Wire
> 
> after "includes"

Ok but where is it?

wergio, 2023-11-10:
config.yaml 
insert at line 12

incarvr6, 2023-11-10:
> config.yaml insert at line 12

Yes but there is no file Wire.h

incarvr6, 2023-11-10:
ok i now manually added wire.h but now new errors:
In file included from src/ABBAurora.cpp:1:
src/ABBAurora.h:2:1: error: 'define' does not name a type
 define ABBAurora_h
 ^~~~~~
*** [.pioenvs/inverter/src/ABBAurora.cpp.o] Error 1

incarvr6, 2023-11-10:
> config.yaml insert at line 12

ok i now manually added wire.h but now new errors:
In file included from src/ABBAurora.cpp:1:
src/ABBAurora.h:2:1: error: 'define' does not name a type
define ABBAurora_h
^~~~~~
*** [.pioenvs/inverter/src/ABBAurora.cpp.o] Error 1

dantheg1t, 2024-05-04:
> > config.yaml insert at line 12
> 
> ok i now manually added wire.h but now new errors: In file included from src/ABBAurora.cpp:1: src/ABBAurora.h:2:1: error: 'define' does not name a type define ABBAurora_h ^~~~~~ *** [.pioenvs/inverter/src/ABBAurora.cpp.o] Error 1

Did you get this resolved? I can't find the wire file.

studiobts, 2024-05-04:
> Did you get this resolved? I can't find the wire file.

You have to add the library to the configuration file, like this:

```
esphome:
  name: Device Name
  ...
  ...
  libraries:
    - "Wire"
```

dantheg1t, 2024-05-04:
Thanks for your help.

I am now getting the following error when installing:-

INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/abbaurora.yaml...
Failed config

libraries: [source /config/esphome/abbaurora.yaml:12]
  
  Component not found: libraries.
  - Wire

wergio, 2024-05-05:
it's not possibile, you are typing something wrong, please follow the documentation on how to include a library in esphome https://esphome.io/components/esphome.html#libraries

dantheg1t, 2024-05-05:
> it's not possibile, you are typing something wrong, please follow the documentation on how to include a library in esphome https://esphome.io/components/esphome.html#libraries
> 
> and remember: in yaml indentation is fundamental

Thank you for helping me resolve this issue. It was the indentation that was wrong.