michelsciortino/esphome-aurora-inverter #19: ESpHome 2025.2.0

leovj, 2025-02-21:
Unfortunately, with the new version of esphome there are two changes to make:
- Please remove the `platform` key from the [esphome] block and use the correct platform component. This style of configuration has now been removed.
-   The "custom" component has been removed. Consider conversion to an external component.

if for the first change the modification could be this:
`esphome:
  name: xxxx
esp32:
  board: esp32doit-devkit-v1`

I have difficulty fixing "custom" components, i.e. using an external component.
Can anyone help me?

bdkacz, 2025-02-24:
Author of component should convert it to  external component

leovj, 2025-02-24:
Yes, I really hope the project can be updated!

leovj, 2025-03-21:
I have found this "temporary" solution:

https://github.com/robertklep/esphome-custom-component

Good Luke! ;)

patrick-mahoney, 2025-03-31:
Thank you to you all for this discussion.   Building on leovj's suggestion, I did as suggested but ran into a few minor issues.

I had to clone the custom component into ./components - otherwise I got an error.

And then a few slight modification to config.yaml:

external_components:
  - source: "./components"
    components: [ custom, custom_component ]

esphome:
  name: inverter
  includes:
    - ABBAuroraStrings.cpp
    - ABBAuroraStrings.h
    - ABBAuroraEnums.h
    - ABBAurora.cpp
    - ABBAurora.h
    - InverterMonitor.h
  libraries:
    - "Wire"

esp32:
  board: platformio_board_id

wifi:
  ssid: SSID
  password: ""

captive_portal:
logger:
api:
ota:
  platform: esphome

<etc>

This built clean for me.

michelsciortino, 2025-05-15:
Hi everyone, feel free to open pull requests and propose your changes :)

patrick-mahoney, 2025-05-25:
I actually did spend a few hours one evening trying to convert the code.  I tried but I'm just not that good a coder.  I forget all this issues that I had but it was an evening of buf whack-a-mole.

In the end though the technique that a few of us used that is outlined above did it get it working for me.

al31c0, 2025-05-25:
@michelsciortino do you have any plan to update the repo?

michelsciortino, 2025-07-24:
Hi everyone, thanks for the attention to my project.

There is this PR opened https://github.com/michelsciortino/esphome-aurora-inverter/pull/21 which aims to convert the code to an external component.

At the moment I don't have direct access to my Inverter so I cannot test directly, If anyone can confirm the PR works, please leave a feedback so I can approve it.